cornell-zhang/allo #145: [Feature] Comment block support

thezenelson, 2024-03-20:
**Problem**
The standard python convention for docstrings requires that it is in the form of a block comment (using """). However, in allo functions comments blocks aren't properly ignored and throw an error.

**Solution**
Support for comment blocks would allow for standard doc strings to work.

chhzh123, 2024-03-20:
Could you provide an example? I didn't quite get the issue here.

chhzh123, 2024-03-20:
You mean you need to use docstring in an Allo kernel? This would need to enhance the parser

thezenelson, 2024-03-21:
[image]
This code will give an unsupported constant type. It would be great if the parser could handle conventional block comments like this. Based on their possession they're automatically treated as docstrings which is a great quality of life feature when using functions/libraries you didn't write/write recently.

thezenelson, 2024-03-21:
This is a min test case that breaks because of the block comment.
```
def min_test():
    def top(x_in: "int8[1]") -> "int8":
        """Test text"""
        return x_in[0]

    print(allo.customize(top).build()(np.array([5], dtype=np.int8)))```

PR #147 (2024-03-25): [Builder] Support comment blocks for Allo kernels

## Description ##
This PR adds support for comment blocks inside the Allo kernels, which should fix #145.

## Checklist ##

- [x] PR's title starts with a category (e.g. [Bugfix], [IR], [Builder], etc)
- [x] Changes are complete (i.e. I finished coding on this PR)
- [x] All changes have test coverage (It would be better to provide ~2 different test cases to test the robustness of your code)
- [x] Code is well-documented
closed 2024-03-25 by chhzh123