cornell-zhang/allo #166: [Feature] Examples of how to import existing HLS module using .compose() primitive

bibo-msft, 2024-07-17:
**Is your feature request related to a problem? Please describe.**
I want to add an existing HLS module into the Allo flow.

**Describe the solution you'd like**
I need an example of how an HLS module, such as [quantize/dequantize](https://github.com/HLSTransform/submission/blob/main/llama_xrt_kernels/src/forward.h) kernels, can be inserted in a design. To be specific, I want to insert an Int8 quantizer before a systolic array and dequantize the result back to Float32/16 as the input to the Gelu operator.

**Additional context**
At the end of Sec.6.1 of the Allo paper, this can be done with .compose() primitive. But I could not find an example in the repo.

chhzh123, 2024-07-23:
@bibo-msft We would appreciate it if you could provide more details about your current project so we can explore how best to support your team. Please feel free to email me at [hzchen@cs.cornell.edu](mailto:hzchen@cs.cornell.edu) to discuss more details or schedule a meeting. Thanks!

YiminGao0113, 2025-04-25:
Hi @chhzh123,
I just wanted to follow up on this to see if there’s any more support now for integrating existing IPs into Allo, such as using an IP as a component within a larger accelerator design (use .compose to integrate the imported HLS module in an accelerator).
Currently, I have a custom bit-serial systolic array HLS kernel that I’d like to integrate into Allo to build a variable-precision quantized inference accelerator.
Thanks a lot!

chhzh123, 2025-04-26:
@YiminGao0113 Yes, we simplify the IP interface a lot, so now you can directly import a C++ HLS kernel as an `allo.IPModule`. Please refer to [this document](https://cornell-zhang.github.io/allo/dive/ip.html) and let us know if it works.