cornell-zhang/allo #171: LLVM patch failed

zeroherolin, 2024-08-13:
I see an error booming out during my installing allo.
It soon works seemly right.
Does it matter?

-----------------------------------------------------------------------
Patching LLVM project ...
error: patch failed: mlir/include/mlir-c/IR.h:826
error: mlir/include/mlir-c/IR.h: patch does not apply
error: patch failed: mlir/lib/Bindings/Python/IRCore.cpp:2277
error: mlir/lib/Bindings/Python/IRCore.cpp: patch does not apply
error: patch failed: mlir/lib/CAPI/IR/IR.cpp:852
error: mlir/lib/CAPI/IR/IR.cpp: patch does not apply
Installing LLVM v18.x ...

chhzh123, 2024-08-13:
Hi, I guess you downloaded the wrong LLVM package. Could you run the tests using the following command and see if Allo is installed correctly?

```cmd
python -m pytest tests
```

zeroherolin, 2024-08-13:
I just run the GEMM test，it works as follows:

----------------------------------------------------------------------------------------------------------------------------------
module {
  func.func @gemm(%arg0: memref<1024x1024xf32>, %arg1: memref<1024x1024xf32>) -> memref<1024x1024xf32> attributes {itypes = "__", otypes = "_"} {
    %alloc = memref.alloc() {name = "C"} : memref<1024x1024xf32>
    %cst = arith.constant 0.000000e+00 : f32
    linalg.fill ins(%cst : f32) outs(%alloc : memref<1024x1024xf32>)
    affine.for %arg2 = 0 to 1024 {
      affine.for %arg3 = 0 to 1024 {
        affine.for %arg4 = 0 to 1024 {
          %0 = affine.load %arg0[%arg2, %arg3] {from = "A"} : memref<1024x1024xf32>
          %1 = affine.load %arg1[%arg3, %arg4] {from = "B"} : memref<1024x1024xf32>
          %2 = arith.mulf %0, %1 : f32
          %3 = affine.load %alloc[%arg2, %arg4] {from = "C"} : memref<1024x1024xf32>
          %4 = arith.addf %3, %2 : f32
          affine.store %4, %alloc[%arg2, %arg4] {to = "C"} : memref<1024x1024xf32>
        } {loop_name = "j"}
      } {loop_name = "k", op_name = "S_k_0", reduction}
    } {loop_name = "i", op_name = "S_i_j_0"}
    return %alloc : memref<1024x1024xf32>
  }
}

module {
  func.func @gemm(%arg0: memref<1024x1024xf32>, %arg1: memref<1024x1024xf32>) -> memref<1024x1024xf32> attributes {itypes = "__", otypes = "_"} {
    %alloc = memref.alloc() {name = "C"} : memref<1024x1024xf32>
    %cst = arith.constant 0.000000e+00 : f32
    linalg.fill ins(%cst : f32) outs(%alloc : memref<1024x1024xf32>)
    affine.for %arg2 = 0 to 1024 {
      %alloc_0 = memref.alloc() : memref<1024xf32>
      affine.for %arg3 = 0 to 1024 {
        affine.store %cst, %alloc_0[%arg3] : memref<1024xf32>
      } {buffer, loop_name = "j_init", pipeline_ii = 1 : i32}
      affine.for %arg3 = 0 to 1024 {
        affine.for %arg4 = 0 to 1024 {
          %0 = affine.load %arg0[%arg2, %arg3] {from = "A"} : memref<1024x1024xf32>
          %1 = affine.load %arg1[%arg3, %arg4] {from = "B"} : memref<1024x1024xf32>
          %2 = arith.mulf %0, %1 : f32
          %3 = affine.load %alloc_0[%arg4] : memref<1024xf32>
          %4 = arith.addf %3, %2 : f32
          affine.store %4, %alloc_0[%arg4] : memref<1024xf32>
        } {loop_name = "j"}
      } {loop_name = "k", op_name = "S_k_0", reduction}
      affine.for %arg3 = 0 to 1024 {
        %0 = affine.load %alloc_0[%arg3] : memref<1024xf32>
        affine.store %0, %alloc[%arg2, %arg3] : memref<1024x1024xf32>
      } {buffer, loop_name = "j_back", pipeline_ii = 1 : i32}
    } {loop_name = "i", op_name = "S_i_j_0"}
    return %alloc : memref<1024x1024xf32>
  }
}

module {
  func.func @gemm(%arg0: memref<1024x1024xf32>, %arg1: memref<1024x1024xf32>) -> memref<1024x1024xf32> attributes {itypes = "__", otypes = "_"} {
    %alloc = memref.alloc() {name = "C"} : memref<1024x1024xf32>
    %cst = arith.constant 0.000000e+00 : f32
    linalg.fill ins(%cst : f32) outs(%alloc : memref<1024x1024xf32>)
    affine.for %arg2 = 0 to 1024 {
      %alloc_0 = memref.alloc() : memref<1024xf32>
      affine.for %arg3 = 0 to 1024 {
        affine.store %cst, %alloc_0[%arg3] : memref<1024xf32>
      } {buffer, loop_name = "j_init", pipeline_ii = 1 : i32}
      affine.for %arg3 = 0 to 1024 {
        affine.for %arg4 = 0 to 1024 {
          %0 = affine.load %arg0[%arg2, %arg3] {from = "A"} : memref<1024x1024xf32>
          %1 = affine.load %arg1[%arg3, %arg4] {from = "B"} : memref<1024x1024xf32>
          %2 = arith.mulf %0, %1 : f32
          %3 = affine.load %alloc_0[%arg4] : memref<1024xf32>
          %4 = arith.addf %3, %2 : f32
          affine.store %4, %alloc_0[%arg4] : memref<1024xf32>
        } {loop_name = "j", pipeline_ii = 1 : i32}
      } {loop_name = "k", op_name = "S_k_0", reduction}
      affine.for %arg3 = 0 to 1024 {
        %0 = affine.load %alloc_0[%arg3] : memref<1024xf32>
        affine.store %0, %alloc[%arg2, %arg3] : memref<1024x1024xf32>
      } {buffer, loop_name = "j_back", pipeline_ii = 1 : i32}
    } {loop_name = "i", op_name = "S_i_j_0"}
    return %alloc : memref<1024x1024xf32>
  }
}

LICENSE file not found
LICENSE file not found
LICENSE file not found
LICENSE file not found
LICENSE file not found

zeroherolin, 2024-08-13:
Except for the impact of 'LICENSE file not found', everything seems normal.

chhzh123, 2024-08-13:
Ok, this looks fine. You can run the entire test suites under the `tests` folder and see if there are other issues.

mxxu00, 2024-08-21:
> i meet same problem. And when i run pytest, it reported:
> 
> ```shell
> _______________________________________ ERROR collecting tests/test_vhls.py ________________________________________
> ImportError while importing test module '/home/mxxu/Desktop/workspace/allo/tests/test_vhls.py'.
> Hint: make sure your test modules/packages have valid Python names.
> Traceback:
> ../../../.conda/envs/allo/lib/python3.12/importlib/__init__.py:90: in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
> tests/test_vhls.py:6: in <module>
>     import allo
> allo/__init__.py:5: in <module>
>     from . import frontend, backend, ir, passes, library
> allo/frontend/__init__.py:4: in <module>
>     from .pytorch import from_pytorch
> allo/frontend/pytorch.py:18: in <module>
>     from .library import CoreAttention_lib, KVCache_lib
> allo/frontend/library.py:6: in <module>
>     from allo.ir.types import float32, int32
> allo/ir/__init__.py:5: in <module>
>     from .types import *
> allo/ir/types.py:8: in <module>
>     from hcl_mlir.ir import IntegerType, IndexType, F16Type, F32Type, F64Type
> E   ModuleNotFoundError: No module named 'hcl_mlir'
> ============================================= short test summary info ==============================================
> ERROR tests/ip_integration/test_external.py
> ERROR tests/pytorch/test_linear.py
> ERROR tests/test_bitop.py
> ERROR tests/test_builder.py
> ERROR tests/test_linalg.py
> ERROR tests/test_monitor_memory.py
> ERROR tests/test_nn.py
> ERROR tests/test_schedule_compose.py
> ERROR tests/test_schedule_compute.py
> ERROR tests/test_schedule_memory.py
> ERROR tests/test_schedule_stream.py
> ERROR tests/test_schedule_structure.py
> ERROR tests/test_systolic_array.py
> ERROR tests/test_template.py
> ERROR tests/test_tensor.py
> ERROR tests/test_types.py
> ERROR tests/test_utils.py
> ERROR tests/test_vhls.py
> !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 18 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
> ================================================ 18 errors in 0.58s ================================================
> ```

it's seem like 'git apply llvm_patch' didn't work, because its reported:
```bash
error: patch failed: mlir/include/mlir-c/IR.h:826
error: mlir/include/mlir-c/IR.h: patch does not apply
error: patch failed: mlir/lib/Bindings/Python/IRCore.cpp:2277
error: mlir/lib/Bindings/Python/IRCore.cpp: patch does not apply
error: patch failed: mlir/lib/CAPI/IR/IR.cpp:852
error: mlir/lib/CAPI/IR/IR.cpp: patch does not apply
```

chhzh123, 2024-08-21:
@mxxu00 It seems your llvm is not correctly installed. Can you check if `llvm-config` work? I'll see what cause the LLVM patch error.

mxxu00, 2024-08-21:
@chhzh123 llvm-config didnt work, just get:
```bash
Command 'llvm-config' not found, but can be installed with:
sudo apt install llvm
```
should i run `apt install llvm`?

mxxu00, 2024-08-21:
@chhzh123 I believe I downloaded the correct LLVM, I specifically checked the version (even went directly to git clone 470625). The installation instructions are exactly according to build.sh, nothing else is added.
Now im trying to use docker way instead of build.sh. 😢

mxxu00, 2024-08-21:
> @mxxu00 Nope, you need to install a specific LLVM (commit [470625](https://github.com/llvm/llvm-project/tree/4706251a3186c34da0ee8fd894f7e6b095da8fdc)). Could you send me the entire installation instructions you followed?

run `git status` at llvm_project folder:

```bash
HEAD detached at 4706251a3186
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   mlir/include/mlir-c/IR.h
	modified:   mlir/lib/Bindings/Python/IRCore.cpp
	modified:   mlir/lib/CAPI/IR/IR.cpp

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	llvm_patch

no changes added to commit (use "git add" and/or "git commit -a")
```

zeroherolin, 2024-08-21:
I meet the very different problem. Most of the functions worked will, but a few of the tests cannot run as normal.
Here's my installation instructions:
```bash
cd ~
git clone https://github.com/cornell-zhang/allo.git --recurse-submodules

cd allo
conda create -n allo python=3.12
conda activate allo

bash build.sh
python3 -m pip install -e .

## fix the error "libstdc++.so.6: GLIBCXX not found" ##
cd ~/anaconda3/envs/allo/lib
sudo rm -f libstdc++.so.6
sudo ln -s /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30 libstdc++.so.6

sudo nano ~/.bashrc
## add follows ##
# export PATH=~/allo/externals/hcl_mlir/externals/llvm-project/build/bin:$PATH
# export XDEVICE='xcvu13p-fhgb2104-2-i'

source ~/.bashrc

cd ~/allo
python -m pytest tests
```
Than 2 of all the tests run into errors showing core dump.
Please check it and give some suggestions.

zeroherolin, 2024-08-21:
> I meet the very different problem. Most of the functions worked will, but a few of the tests cannot run as normal. Here's my installation instructions:
> 
> ```shell
> cd ~
> git clone https://github.com/cornell-zhang/allo.git --recurse-submodules
> 
> cd allo
> conda create -n allo python=3.12
> conda activate allo
> 
> bash build.sh
> python3 -m pip install -e .
> 
> ## fix the error "libstdc++.so.6: GLIBCXX not found" ##
> cd ~/anaconda3/envs/allo/lib
> sudo rm -f libstdc++.so.6
> sudo ln -s /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30 libstdc++.so.6
> 
> sudo nano ~/.bashrc
> ## add follows ##
> # export PATH=~/allo/externals/hcl_mlir/externals/llvm-project/build/bin:$PATH
> # export XDEVICE='xcvu13p-fhgb2104-2-i'
> 
> source ~/.bashrc
> 
> cd ~/allo
> python -m pytest tests
> ```
> 
> Than 2 of all the tests run into errors showing core dump. Please check it and give some suggestions.

Here's the "core dump" issue.
https://github.com/cornell-zhang/allo/issues/172#issue-2464319753

mxxu00, 2024-09-09:
@chhzh123 So grateful for your reply. This situation make me confused. But nevermind, i have successsfully used Docker and Allo can run correctly!

chhzh123, 2024-09-09:
Great! I'll close this issue.

closed 2024-09-09 by chhzh123