cornell-zhang/allo #176: [BUG] NaN Wrong Results of Scaled Dot Product Attention due to Softmax Overflow

silvenachen, 2024-09-09:
**Describe the bug**
When I'm calling scaled_dot_product_attention() function from Allo pre-build library in a larger kernel, the intermediate results are incorrect and yields NaNs. The issue stems from numerical overflow during the softmax computation, where large values in the exponentiation step cause overflow, leading to NaNs.

**To Reproduce**
```
import allo
from allo.ir.types import float32
import numpy as np

def test_allo_sdp():
    from allo.library.nn import scaled_dot_product_attention
    L, D = 1024, 512
    Q = np.random.randn(L, D).astype(np.float32)
    K = np.random.randn(L, D).astype(np.float32)
    V = np.random.randn(L, D).astype(np.float32)
    Q = 1000*Q
    K = 1000*K
    V = 1000*V
    H = 8
    Q_np = Q.reshape((L, H, D//H))
    K_np = K.reshape((L, H, D//H))
    V_np = V.reshape((L, H, D//H))

    s = allo.customize(scaled_dot_product_attention, instantiate=[float32, H, L, D, 8, 8])
    mod = s.build()
    allo_out = mod(Q, K, V)
    print(allo_out)
```

if __name__ == "__main__":
    test_allo_sdp()

**Buggy output**
[[nan nan nan ... nan nan nan]
 [nan nan nan ... nan nan nan]
 [nan nan nan ... nan nan nan]
 ...
 [nan nan nan ... nan nan nan]
 [nan nan nan ... nan nan nan]
 [nan nan nan ... nan nan nan]]

**Expected behavior**
The scaled_dot_product_attention should function and gives correct numerical output even when input is large.

**Additional context**
The cause of NaNs is due to the overflow of softmax operation, as the original Allo function can not handle larger input numbers and results in overflow.

chhzh123, 2024-09-09:
@silvenachen Thanks for raising this issue! Have you fixed the softmax function? If so, could you help create a PR for Allo?

silvenachen, 2024-09-09:
> @silvenachen Thanks for raising this issue! Have you fixed the softmax function? If so, could you help create a PR for Allo?

Yes, I think I have fixed this issue for both softmax and attention computation. How could I help to maybe update the library?

closed 2024-09-10 by chhzh123