cornell-zhang/allo #241: [BUG] "Casting between unsupported types" error for fixed type and index type

cshi15, 2024-11-01:
**Describe the bug**
I'm trying to write something that uses a Fixed(32, 24) type and a Fixed(64, 48) type for data. In a loop, I have a variable step that increments in the loop and is of type Index. I'm trying to perform an operation (add) with the Fixed(32, 24) type and the Index type, and store the result in the Fixed(64, 48) variable. However, I get the error "[Data Type] Casting between unsupported types. src type: index, dst type: fixed(58, 24)". I tried changing the type of the destination variable to Fixed(58, 24), but that resulted in the same error, so I'm guessing that the issue is in the add operation between the Fixed(32, 24) type and the Index type. However, this doesn't seem to strictly be an add issue either, since originally it was a shift operation, but that still gave the same error.

**To Reproduce**
```
import allo                                                                                                             
from allo.ir.types import int32, bool, index, Fixed, float32                                                                               

theta_type = Fixed(32, 24)                                             
mult_type = Fixed(64, 48)                                                                    

def test(theta: theta_type):                                                                                                
    current_sin: theta_type = 0.0                                                                     
    for step in range(20):         
        mult_intermed_cos: mult_type = 0.0                                 
        mult_intermed_cos = current_sin + step 

s = allo.customize(test)       
print(s.module)                                          
mod = s.build(target="llvm")     
```

**Buggy output**
raise DTypeError(allo._mlir.exceptions.DTypeError: [Data Type] Casting between unsupported types. src type: index, dst type: fixed(58, 24)

**Expected behavior**
Expect the program to run correctly, and for the types to automatically cast so that the correct result is stored in mult_intermed_cos with the Fixed(64,48) datatype.

chhzh123, 2024-11-01:
Hi @cshi15, thanks for raising the issue. Currently there is no direct casting from index type to fixed type in Allo, which will be added later. A temporary workaround is to explicitly cast `step` to integer type, i.e., `int(step)`

cshi15, 2024-11-01:
Thank you, that works for the add operation! However, when I changed it back to the shift operation (<<), I got the following `RuntimeError: Typing rule is not defined with input types (<class 'allo.ir.types.Int'>, <class 'allo.ir.types.Fixed'>)`. If I cast current_sin as an int, it runs, but I'm not sure if casting this way would lead to incorrect behavior? I'm generally confused about when I need to cast and when I don't, and if it's possible to cast something as an Allo dtype (for example, casting something as Fixed)

chhzh123, 2024-11-03:
Hi @cshi15, this is not a very common use case. In Allo, we suppose both operands of the `<<` are integers. Can you tell us what you plan to do here?

In general, Allo does not require users to write explicit data casting as long as there exists a valid casting rule. We mostly follow the C/C++ convention to construct the casting rules. If some types are not supported by the operation, then Allo will error out just like this case.

andrewb1999, 2024-11-03:
@cshi15 wants to use this for a cordic core so the `val >> n` is essentially shorthand for dividing by 2^n. This is something Vitis HLS supports but I understand if you don't want to allow this. Will allo do strength reduction for dividing by 2^n?

chhzh123, 2024-11-03:
@andrewb1999 thanks for clarifying! I just implemented the fixed point shifting operations in Allo (#243). @cshi15 Please pull the latest commit, reinstall Allo by `pip install -v -e .`, and try again for this case. The generated module is listed below for your reference:

```mlir
module {
  func.func @test(%arg0: !allo.Fixed<32, 24>) attributes {itypes = "s", otypes = ""} {
    %cst = arith.constant 0.000000e+00 : f32
    %0 = allo.float_to_fixed(%cst) : f32 -> !allo.Fixed<32, 24>
    %alloc = memref.alloc() {name = "current_sin"} : memref<!allo.Fixed<32, 24>>
    affine.store %0, %alloc[] {to = "current_sin"} : memref<!allo.Fixed<32, 24>>
    affine.for %arg1 = 0 to 20 {
      %cst_0 = arith.constant 0.000000e+00 : f32
      %1 = allo.float_to_fixed(%cst_0) : f32 -> !allo.Fixed<64, 48>
      %alloc_1 = memref.alloc() {name = "mult_intermed_cos"} : memref<!allo.Fixed<64, 48>>
      affine.store %1, %alloc_1[] {to = "mult_intermed_cos"} : memref<!allo.Fixed<64, 48>>
      %2 = arith.index_cast %arg1 : index to i32
      %3 = affine.load %alloc[] {from = "current_sin"} : memref<!allo.Fixed<32, 24>>
      %4 = allo.shl_fixed(%3, %2) :(!allo.Fixed<32, 24>, i32) -> !allo.Fixed<32, 24>
      %5 = allo.fixed_to_fixed(%4) : !allo.Fixed<32, 24> -> !allo.Fixed<64, 48>
      affine.store %5, %alloc_1[] {to = "mult_intermed_cos"} : memref<!allo.Fixed<64, 48>>
    } {loop_name = "step", op_name = "S_step_0"}
    return
  }
}
```

Feel free to comment if there are additional issues.

PR #243 (2024-11-03): [IR] Add shift left/right operations for fixed points

## Description ##
This PR tries to fix #241 by adding new shifting operations for fixed types. Reference example comes from the official [documentation](https://docs.amd.com/r/en-US/ug1399-vitis-hls/Class-Methods-Operators-and-Data-Members) of `ap_int`.

```python
def test_fixed_shift():
    # Example from 
    def kernel(A: Fixed(8, 3)[2]) -> float32[2]:
        shl: Fixed(25, 10) = A[0] << 2
        shr: Fixed(25, 10) = A[1] >> 2
        res: float32[2] = 0.0
        res[0] = float(shl)
        res[1] = float(shr)
        return res

    [... cut]
closed 2024-11-03 by chhzh123