cornell-zhang/allo #261: [BUG] Struct Usage

cshi15, 2024-11-20:
**Describe the bug**
When I try to use a struct, it doesn't seem to work and gives an error. I took a look at the implementation for Struct, and though it seems partially implemented I saw that there is a "TODO" under the build function. I just wanted to check if structs can actually be used yet, and if so, if I am using them correctly? My code works if I take out the struct and pass in all the arguments individually. I'm getting the following error:
```RuntimeError: Can only access bit (slice) for integers```

**To Reproduce**
Here is the code I am running:

```
import allo  
from allo.ir.types import int3, int4, int16, int64, bool, index, Fixed, Struct

xcel_req_t = Struct({"opcode": int4, "addr_d": int64, "len_d": int16, "addr_1": int64, "len_1": int16, "addr_2": int64, "len_2": int16, "dtype": int3, "acc": bool})

def alu_top(xcel_reqstream_msg: xcel_req_t, xcel_reqstream_rdy: bool, xcel_reqstream_val: bool, arr_in_1: int64[1000], arr_in_2: int64[1000]) -> int64[1000]:

    alu_func: int4 = xcel_reqstream_msg["opcode"]
    addr_1_dec: int64 = xcel_reqstream_msg["addr_1"]
    addr_2_dec: int64 = xcel_reqstream_msg["addr_2"]
    addr_d_dec: int64 = xcel_reqstream_msg["addr_d"]

    arr_len_1: int16 = xcel_reqstream_msg["len_1"]
    arr_len_2: int16 = xcel_reqstream_msg["len_2"]
    arr_len_d: int16 = xcel_reqstream_msg["len_d"]

    arr_out: int64[1000] = 0
    for i in range(len_1):
        if alu_func == 0:
            arr_out[addr_d_dec] = arr_in_1[addr_1_dec] + arr_in_2[addr_2_dec]
        if alu_func == 1:
            arr_out[addr_d_dec] = arr_in_1[addr_1_dec] - arr_in_2[addr_2_dec]
        if alu_func == 2:
            arr_out[addr_d_dec] = arr_in_1[addr_1_dec] * arr_in_2[addr_2_dec]
        addr_1_dec = addr_1_dec + 1
        addr_2_dec = addr_2_dec + 1
        addr_d_dec = addr_d_dec + 1
                                  
    return arr_out

s = allo.customize(alu_top)
print(s.module)
```

**Buggy output**
```
Traceback (most recent call last):
  File "/home/cys37/alu_top.py", line 41, in <module>
    s = allo.customize(alu_top)
        ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/cys37/allo/allo/customize.py", line 916, in customize
    tree = TypeInferer()(ctx_type_inf, tree)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/cys37/allo/allo/ir/visitor.py", line 111, in __call__
    res = method(ctx, node)
          ^^^^^^^^^^^^^^^^^
  File "/home/cys37/allo/allo/ir/infer.py", line 653, in visit_Module
    visit_stmt(ctx, stmt)
  File "/home/cys37/allo/allo/ir/visitor.py", line 111, in __call__
    res = method(ctx, node)
          ^^^^^^^^^^^^^^^^^
  File "/home/cys37/allo/allo/ir/infer.py", line 582, in visit_FunctionDef
    visit_stmts(ctx, node.body)
  File "/home/cys37/allo/allo/ir/infer.py", line 1007, in visit_stmts
    raise e
  File "/home/cys37/allo/allo/ir/infer.py", line 1005, in visit_stmts
    results.append(visit_stmt(ctx, stmt))
                   ^^^^^^^^^^^^^^^^^^^^^
  File "/home/cys37/allo/allo/ir/visitor.py", line 111, in __call__
    res = method(ctx, node)
          ^^^^^^^^^^^^^^^^^
  File "/home/cys37/allo/allo/ir/infer.py", line 521, in visit_AnnAssign
    visit_stmt(ctx, node.value)
  File "/home/cys37/allo/allo/ir/visitor.py", line 111, in __call__
    res = method(ctx, node)
          ^^^^^^^^^^^^^^^^^
  File "/home/cys37/allo/allo/ir/infer.py", line 465, in visit_Subscript
    raise RuntimeError("Can only access bit (slice) for integers")
RuntimeError: Can only access bit (slice) for integers
```
**Expected behavior**
I expect it to print out the IR of the module.

zzzDavid, 2024-11-20:
I see that we have the frontend support for `Struct` interface but the `build()` function is not implemented yet. I will implement it after ISCA submission.
https://github.com/cornell-zhang/allo/blob/7a84a7bcb24c96d0ceca57ae74bd376a40bd98f8/allo/ir/types.py#L235-L236

chhzh123, 2024-11-26:
Hi @cshi15, please pull the latest commit (#265) and check if it works. A struct example is [here](https://github.com/cornell-zhang/allo/blob/main/tests/test_types.py#L509-L539). Remember to rebuild Allo using `pip install -v -e .` after you pull it.

chhzh123, 2024-12-12:
The generated module is listed below, so I'll close this issue.

```mlir
module {
  func.func @alu_top(%arg0: !allo.struct<i4, i64, i16, i64, i16, i64, i16, i3, i1>, %arg1: i1, %arg2: i1, %arg3: memref<1000xi64>, %arg4: memref<1000xi64>) -> memref<1000xi64> attributes {itypes = "_uuss", otypes = "s"} {
    %0 = allo.struct_get %arg0[0] : <i4, i64, i16, i64, i16, i64, i16, i3, i1> -> i4
    %alloc = memref.alloc() {name = "alu_func"} : memref<i4>
    affine.store %0, %alloc[] {to = "alu_func"} : memref<i4>
    %1 = allo.struct_get %arg0[3] : <i4, i64, i16, i64, i16, i64, i16, i3, i1> -> i64
    %alloc_0 = memref.alloc() {name = "addr_1_dec"} : memref<i64>
    affine.store %1, %alloc_0[] {to = "addr_1_dec"} : memref<i64>
    %2 = allo.struct_get %arg0[5] : <i4, i64, i16, i64, i16, i64, i16, i3, i1> -> i64
    %alloc_1 = memref.alloc() {name = "addr_2_dec"} : memref<i64>
    affine.store %2, %alloc_1[] {to = "addr_2_dec"} : memref<i64>
    %3 = allo.struct_get %arg0[1] : <i4, i64, i16, i64, i16, i64, i16, i3, i1> -> i64
    %alloc_2 = memref.alloc() {name = "addr_d_dec"} : memref<i64>
    affine.store %3, %alloc_2[] {to = "addr_d_dec"} : memref<i64>
    %4 = allo.struct_get %arg0[4] : <i4, i64, i16, i64, i16, i64, i16, i3, i1> -> i16
    %alloc_3 = memref.alloc() {name = "arr_len_1"} : memref<i16>
    affine.store %4, %alloc_3[] {to = "arr_len_1"} : memref<i16>
    %5 = allo.struct_get %arg0[6] : <i4, i64, i16, i64, i16, i64, i16, i3, i1> -> i16
    %alloc_4 = memref.alloc() {name = "arr_len_2"} : memref<i16>
    affine.store %5, %alloc_4[] {to = "arr_len_2"} : memref<i16>
    %6 = allo.struct_get %arg0[2] : <i4, i64, i16, i64, i16, i64, i16, i3, i1> -> i16
    %alloc_5 = memref.alloc() {name = "arr_len_d"} : memref<i16>
    affine.store %6, %alloc_5[] {to = "arr_len_d"} : memref<i16>
    %c0_i32 = arith.constant 0 : i32
    %7 = arith.extsi %c0_i32 : i32 to i64
    %alloc_6 = memref.alloc() {name = "arr_out"} : memref<1000xi64>
    linalg.fill ins(%7 : i64) outs(%alloc_6 : memref<1000xi64>)
    affine.for %arg5 = 0 to 10 {
      %8 = affine.load %alloc[] {from = "alu_func"} : memref<i4>
      %9 = arith.extsi %8 : i4 to i32
      %c0_i32_7 = arith.constant 0 : i32
      %10 = arith.cmpi eq, %9, %c0_i32_7 : i32
      scf.if %10 {
        %32 = affine.load %alloc_0[] {from = "addr_1_dec"} : memref<i64>
        %33 = arith.index_cast %32 : i64 to index
        %34 = memref.load %arg3[%33] {from = "arr_in_1"} : memref<1000xi64>
        %35 = affine.load %alloc_1[] {from = "addr_2_dec"} : memref<i64>
        %36 = arith.index_cast %35 : i64 to index
        %37 = memref.load %arg4[%36] {from = "arr_in_2"} : memref<1000xi64>
        %38 = arith.extsi %34 : i64 to i65
        %39 = arith.extsi %37 : i64 to i65
        %40 = arith.addi %38, %39 : i65
        %41 = arith.trunci %40 : i65 to i64
        %42 = affine.load %alloc_2[] {from = "addr_d_dec"} : memref<i64>
        %43 = arith.index_cast %42 : i64 to index
        memref.store %41, %alloc_6[%43] {to = "arr_out"} : memref<1000xi64>
      }
      %11 = affine.load %alloc[] {from = "alu_func"} : memref<i4>
      %12 = arith.extsi %11 : i4 to i32
      %c1_i32 = arith.constant 1 : i32
      %13 = arith.cmpi eq, %12, %c1_i32 : i32
      scf.if %13 {
        %32 = affine.load %alloc_0[] {from = "addr_1_dec"} : memref<i64>
        %33 = arith.index_cast %32 : i64 to index
        %34 = memref.load %arg3[%33] {from = "arr_in_1"} : memref<1000xi64>
        %35 = affine.load %alloc_1[] {from = "addr_2_dec"} : memref<i64>
        %36 = arith.index_cast %35 : i64 to index
        %37 = memref.load %arg4[%36] {from = "arr_in_2"} : memref<1000xi64>
        %38 = arith.extsi %34 : i64 to i65
        %39 = arith.extsi %37 : i64 to i65
        %40 = arith.subi %38, %39 : i65
        %41 = arith.trunci %40 : i65 to i64
        %42 = affine.load %alloc_2[] {from = "addr_d_dec"} : memref<i64>
        %43 = arith.index_cast %42 : i64 to index
        memref.store %41, %alloc_6[%43] {to = "arr_out"} : memref<1000xi64>
      }
      %14 = affine.load %alloc[] {from = "alu_func"} : memref<i4>
[... 19 lines cut ...]
      %18 = arith.extsi %17 : i64 to i65
      %c1_i32_8 = arith.constant 1 : i32
      %19 = arith.extsi %c1_i32_8 : i32 to i65
      %20 = arith.addi %18, %19 : i65
      %21 = arith.trunci %20 : i65 to i64
      affine.store %21, %alloc_0[] {to = "addr_1_dec"} : memref<i64>
      %22 = affine.load %alloc_1[] {from = "addr_2_dec"} : memref<i64>
      %23 = arith.extsi %22 : i64 to i65
      %c1_i32_9 = arith.constant 1 : i32
      %24 = arith.extsi %c1_i32_9 : i32 to i65
      %25 = arith.addi %23, %24 : i65
      %26 = arith.trunci %25 : i65 to i64
      affine.store %26, %alloc_1[] {to = "addr_2_dec"} : memref<i64>
      %27 = affine.load %alloc_2[] {from = "addr_d_dec"} : memref<i64>
      %28 = arith.extsi %27 : i64 to i65
      %c1_i32_10 = arith.constant 1 : i32
      %29 = arith.extsi %c1_i32_10 : i32 to i65
      %30 = arith.addi %28, %29 : i65
      %31 = arith.trunci %30 : i65 to i64
      affine.store %31, %alloc_2[] {to = "addr_d_dec"} : memref<i64>
    } {loop_name = "i", op_name = "S_i_0"}
    return %alloc_6 : memref<1000xi64>
  }
}
```

closed 2024-12-12 by chhzh123