cornell-zhang/allo #278: [BUG] Windows compilation and run

gelonsoft, 2024-12-15:
**Describe the bug**
Can't compile and run without changes

**To Reproduce**
See instruction to compile from source from official Allo manual

**Buggy output**
Errors on compilation

**Expected behavior**
Instruction for windows compilation that's running without errors

**Additional context**
Successfully compiled and tested with the following instruction and changes:

!Install Visual Studio Community 2019 lates with C++ ATL MFC support. For me it's in c:\Program Files (x86)\Microsoft Visual Studio\2019\Community
!Download from github allo.zip and unpack to e:\dev\allo (for me here, but you can use any dir and change path in instruction below)
!Download from github llvm-project.zip and unpack to e:\dev\allo\externals\llvm-project (for me allo in e:\dev\allo)
!Download and unpack ninja.exe to e:\dev\ninja (for me here, but you can use any dir and change path in instruction below)
!Download and unpack cmake to e:\dev\CMake (for me here, but you can use any dir and change path in instruction below)
!Install Python 3.12 to e:\dev\Python312 (for me here, but you can use any dir and change path in instruction below)

"c:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Auxiliary\Build\vcvarsall.bat" x64
cd e:\dev\allo\externals\llvm-project\
e:\dev\Python312\python.exe -m pip install -r e:\dev\allo\externals\llvm-project\mlir\python\requirements.txt
mkdir build && cd build
e:\dev\CMake\bin\cmake.exe -G Ninja ../llvm    -DCMAKE_MAKE_PROGRAM=e:\dev\ninja\ninja.exe  -DLLVM_ENABLE_PROJECTS=mlir       -DLLVM_BUILD_EXAMPLES=ON       -DLLVM_TARGETS_TO_BUILD="host"       -DCMAKE_BUILD_TYPE=Release       -DLLVM_ENABLE_ASSERTIONS=ON       -DLLVM_INSTALL_UTILS=ON       -DMLIR_ENABLE_BINDINGS_PYTHON=ON  -DPython3_EXECUTABLE=e:\dev\Python312\python.exe  
e:\dev\ninja\ninja.exe
set LLVM_BUILD_DIR=e:\dev\allo\externals\llvm-project\build2

cd e:\dev\allo\
e:\dev\Python312\python.exe -m pip install -r requirements.txt
vi E:\dev\allo\mlir\include\allo\Bindings\CMakeLists.txt
-------
  ROOT_DIR "${ALLO_PYTHON_SOURCE_DIR}"
  SOURCES
    AlloModule.cpp
    AlloTypes.cpp
    AlloAttributes.cpp
------
vi E:\dev\allo\mlir\CMakeLists.txt
-----
<...>
set(CMAKE_CXX_STANDARD 17 CACHE STRING "C++ standard to conform to")
#CHANGE START
set(CMAKE_CXX_FLAGS " /std=c++17")
add_compile_options ( -w )

#set(CMAKE_BUILD_TYPE Debug)
set(CMAKE_BUILD_TYPE Release)
#CHANGE END
find_package(MLIR REQUIRED CONFIG)
<...>
----
vi E:\dev\allo\mlir\lib\Transforms\CMakeLists.txt
-----
<...>
    LINK_LIBS PUBLIC
    MLIRIR
    MLIRPass
    MLIRAllo
	#ADDED BEGIN
	MLIRAlloConversion
	#ADDED DONE
    MLIRAlloSupport
<...>
------
cmake -G Ninja E:\dev\allo\mlir -DMLIR_DIR=e:\dev\allo\externals\llvm-project\build2\lib\cmake\mlir -DPython3_EXECUTABLE=e:\dev\Python312\python.exe -Wno-dev 
e:\dev\ninja\ninja.exe

cd e:\dev\allo\allo\
!copy ..\mlir\build\tools\allo\_mlir to _mlir
cd e:\dev\allo
vi setup.py
------
        #subprocess.run(
        #    ["cmake", "-G Ninja", ext.sourcedir] + cmake_args,
        #    cwd=build_temp,
        #    check=True,
        #)
        #subprocess.run(["ninja"], cwd=build_temp, check=True)
--------
e:\dev\Python312\python.exe -m pip install -v -e .
vi test.py
--------
#ADDED TO BEGIN:
import os
os.add_dll_directory('e:/dev/allo/allo/_mlir')
#ADDED DONE
import allo as allo
import allo.ir as air
-----------
e:\dev\Python312\python.exe test.py
vi examples/transformer_hls.py
----
#ADDED TO BEGIN:
import os
os.add_dll_directory('e:/dev/allo/allo/_mlir')
#ADDED DONE
<other file>
-----
e:\dev\Python312\python.exe examples/transformer_hls.py

zzzDavid, 2024-12-15:
Hi, if you would like to use Allo on a Windows PC, I suggest to try the "Install from Docker" option: https://cornell-zhang.github.io/allo/setup/index.html#install-from-docker

chhzh123, 2025-01-03:
Windows [WSL](https://learn.microsoft.com/en-us/windows/wsl/install) may be another option for running Allo. Please reopen the issue if the problem still exists.

closed 2025-01-03 by chhzh123