cornell-zhang/allo #279: [BUG] ValueError: DenseElementsAttr.get np_values.dtype=int32, type.build=i64

gelonsoft, 2024-12-15:
**Describe the bug**
Values before line value_attr = DenseElementsAttr.get(np_values, type=dtype.build()) in  File ALLO_DIR\allo\ir\builder.py, line 844:
np_values=[   2 1024   12   64], np_values.dtype=int32, type.build=i64

Traceback (most recent call last):
  File "E:\dev\allo\allo\ir\builder.py", line 2401, in build_stmts
    results.append(build_stmt(ctx, stmt))
                   ^^^^^^^^^^^^^^^^^^^^^
  File "E:\dev\allo\allo\ir\builder.py", line 76, in __call__
    res = method(ctx, node, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\dev\allo\allo\ir\builder.py", line 786, in build_Assign
    rhs = build_stmt(ctx, node.value)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\dev\allo\allo\ir\builder.py", line 76, in __call__
    res = method(ctx, node, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\dev\allo\allo\ir\builder.py", line 1918, in build_Call
    return ASTTransformer.build_library_op(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\dev\allo\allo\ir\builder.py", line 2203, in build_library_op
    shape_value = ASTTransformer.build_constant_tensor(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\dev\allo\allo\ir\builder.py", line 844, in build_constant_tensor
    value_attr = DenseElementsAttr.get(np_values, type=dtype.build())
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: DenseElementsAttr could not be constructed from the given buffer. This may mean that the Python buffer layout does not match that MLIR expected layout and is a bug.

Error: DenseElementsAttr could not be constructed from the given buffer. This may mean that the Python buffer layout does
not match that MLIR expected layout and is a bug.
+------------------------------------ Traceback (most recent call last): Line: 199 -------------------------------------+
|  194:     ln_f_weight: float32[768] = g_ln_f_weight                                                                   |
|  195:     ln_f_bias: float32[768] = g_ln_f_bias                                                                       |
|  196:     fc_weight: float32[50257, 768] = g_fc_weight                                                                |
|  197:     fc_bias: float32[50257] = g_fc_bias                                                                         |
|  198:     transformer_blocks_0_attention_linear_q = dsl.linear(x, transformer_blocks_0_attention_linear_q_weight,     |
| transformer_blocks_0_attention_linear_q_bias)                                                                         |
|  199:     view = dsl.view(transformer_blocks_0_attention_linear_q, (2, 1024, 12, 64))                                 |
|  200:     permute = dsl.transpose(view, (0, 2, 1, 3))                                                                 |
|  201:     transformer_blocks_0_attention_linear_k = dsl.linear(x, transformer_blocks_0_attention_linear_k_weight,     |
| transformer_blocks_0_attention_linear_k_bias)                                                                         |
|  202:     view_1 = dsl.view(transformer_blocks_0_attention_linear_k, (2, 1024, 12, 64))                               |
|  203:     permute_1 = dsl.transpose(view_1, (0, 2, 1, 3))                                                             |
+----------------------------------------------------- Source Code -----------------------------------------------------+

**To Reproduce**
1. Install allo
2. Run examples/torch/gpt.py

**Buggy output**
See above

**Expected behavior**
No errors, success compilation

**Additional context**
OS: Windows
allo: 0.3 (compiled from source using Visual Studio 2019 Community) 
Python 3.12.8
torch[cpu]: 2.5.1
numpy: 1.26.0

zzzDavid, 2024-12-15:
Hello, what is the LLVM version in your environment? Were you able to run the tests [here](https://cornell-zhang.github.io/allo/setup/index.html#testing) to confirm you have a working Allo installation?

chhzh123, 2025-01-03:
Please make sure you pull and install the latest commit of Allo. This issue will be closed due to no further updates.

closed 2025-01-03 by chhzh123