cornell-zhang/allo #298: [BUG] The dataflow testcase `test_multicache_gemm` generates invalid MLIR module

Oh-Afterglow, 2025-02-13:
**Describe the bug**
Building the kernel in the multi-cache GEMM case results in invalid MLIR module. If printed the module will be in the generic format instead of the pretty format. Parsing the module as string with `Module.parse()` will cause an `Unable to parse module assembly` error.

**To Reproduce**
Use the build statement provided in the testcase and print the module generated:
```python
mod1 = df.build(top, wrap_io=False)
print(mod1.module)
```
Try parsing the module:
```python
with Context() as ctx:
    module = Module.parse(str(mod1.module), ctx)
```

**Buggy output**
Part of the printed MLIR:
```mlir
"builtin.module"() ({
  "func.func"() <{function_type = (memref<2048xi64>, !allo.stream<i64, 4>) -> (), sym_name = "offchip_loadA_0"}> ({
  ^bb0(%arg788: memref<2048xi64>, %arg789: !allo.stream<i64, 4>):
    "affine.for"() <{lowerBoundMap = #map1, operandSegmentSizes = array<i32: 0, 0, 0>, step = 1 : index, upperBoundMap = #map3}> ({
    ^bb0(%arg790: index):
      "affine.for"() <{lowerBoundMap = #map1, operandSegmentSizes = array<i32: 0, 0, 0>, step = 1 : index, upperBoundMap = #map3}> ({
      ^bb0(%arg791: index):
        "affine.for"() <{lowerBoundMap = #map1, operandSegmentSizes = array<i32: 0, 0, 0>, step = 1 : index, upperBoundMap = #map2}> ({
        ^bb0(%arg792: index):
          %3353 = "affine.load"(%arg788, %arg790, %arg792) <{map = #map}> {from = "A_Packed", unsigned} : (memref<2048xi64>, index, index) -> ui64
          "allo.stream_put"(%arg789, %3353) <{indices = array<i64>}> : (!allo.stream<i64, 4>, ui64) -> ()
          "affine.yield"() : () -> ()
        }) {loop_name = "k", op_name = "S_k_0", pipeline_ii = 1 : ui32, rewind} : () -> ()
        "affine.yield"() : () -> ()
      }) {loop_name = "nt"} : () -> ()
      "affine.yield"() : () -> ()
    }) {loop_name = "mt", op_name = "S_mt_nt_0"} : () -> ()
    "func.return"() : () -> ()
  }) {stypes = "_o"} : () -> ()
...
```
The prompted error when parsing:
```
allo._mlir._mlir_libs._site_initialize.<locals>.MLIRError: Unable to parse module assembly:
error: "-":24:19: 'affine.load' op result type must match element type of memref
 note: "-":24:19: see current operation: %0 = "affine.load"(%arg0, %arg2, %arg4) <{map = affine_map<(d0, d1) -> (d0 * 128 + d1)>}> {from = "A_Packed", unsigned} : (memref<2048xi64>, index, index) -> ui64
error: "-":42:19: 'affine.load' op result type must match element type of memref
 note: "-":42:19: see current operation: %0 = "affine.load"(%arg0, %arg3, %arg4) <{map = affine_map<(d0, d1) -> (d0 * 128 + d1)>}> {from = "B_Packed", unsigned} : (memref<2048xi64>, index, index) -> ui64
```

**Expected behavior**
Building the module results in a valid MLIR module so that it can be transformed into the kernel's dataflow simulator by other MLIR passes.

**Additional context**
When `build()` is called for the kernel, a warning message is given by NumPy:
```
[Data Type] NumPy does not support bitwidths larger than 512, so you cannot use NumPy for simulation, but you can still generate the corresponding MLIR module.
```

AdrianLiu00, 2025-02-14:
It looks related to the type mismatch of the load operation from packed data, such as: `L3_A.put(A_Packed[mt * K + k])`.
And there are two questions to answer or address:
1. `A_Packed` is declared as **UInt**(64)[2048], but the prompted error shows it's actually **i64** for the memref its elements. (This is probably caused by the weak support for unsigned type in MLIR);
2.  The return value built for Load operation is **ui64**. I'm not sure how it works and whether it will lead to some mismatch problem, especially under the aforementioned circumstance.

I hope above thoughts could offer some clues for this issue.

chhzh123, 2025-02-20:
@Oh-Afterglow The invalid MLIR module is because you access the one *after* the HLS codegen. Your `mod1` here is actually an HLSModule instead of the original target-independent module, so the signedness information has been injected after the codegen. Your simulator should take the MLIR module immediately after the `customize` function.
https://github.com/cornell-zhang/allo/blob/main/allo/dataflow.py#L269

PR #302 (2025-02-21): [dataflow] Support UInt stream

## Description ##
This PR adds unsigned type annotation to the stream construct. This PR fixes #298.

## Checklist ##

- [x] PR's title starts with a category (e.g. [Bugfix], [IR], [Builder], etc)
- [x] Changes are complete (i.e. I finished coding on this PR)
- [x] All changes have test coverage (It would be better to provide ~2 different test cases to test the robustness of your code)
- [x] Code is well-documented
closed 2025-02-21 by chhzh123