cornell-zhang/allo #311: C++ complilation

Chubbysoap, 2025-03-03:
[image]  

After running 'python3 -m pip install -v -e .' Get 'c++: fatal error: Killed signal terminated program cc1plus
    compilation terminated.' When it is finished running I get ' × python setup.py develop did not run successfully.
    │ exit code: 1
    ╰─> See above for output.'

I have 8Gb allocated to docker.

chhzh123, 2025-03-03:
This is because you extensively used the CPU cores for compilation. Please check the number of CPU cores on your machine, and explicitly set the number of threads for compilation using `NUM_THREADS=4 python3 -m pip install -v -e .`.

Chubbysoap, 2025-03-04:
Worked with 'NUM_THREADS=6'.  Thanks.

closed 2025-03-05 by chhzh123