cornell-zhang/allo #322: [BUG] The result type of the allo.dsl function was incorrectly inferred

zsiothsu, 2025-03-23:
**Describe the bug**
The result type of the allo.dsl function was incorrectly inferred. I encountered an error when using allo.dsl.add to add a tensor and a scalar. When the `lhs` is a tensor and the `rhs` is a scalar, the result is correct. However, when the `lhs` is a scalar and the `rhs` is a tensor, an error occurs.

allo.dsl中的函数类型被错误推导，比如 allo.dsl.add.函数，当 lhs为张量、rhs标量时，一切正常。但是交换 lhs 和 rhs, lhs 为标量、rhs为向量时，类型被错误推导

The bug was caused by the code in allo/ir/infer.py:TypeInfer.visit_Call
This code incorrectly identifies the add operation as an element-wise computation.

这个bug由 allo/ir/infer.py中的TypeInfer.visiti_Call函数引发，下面这段代码仅判断了第一个参数，将函数误认为了element-wise操作：

```python
# allo/ir/infer.py

if len(new_args[0].shape) == 0:
    # element-wise operation
    node.shape = tuple()
    node.dtype = new_args[0].dtype
    return node
```

**To Reproduce**

```python
def foo(x: int32[2,3]) -> int32[2,3]:
    return allo.dsl.add(1, x)

s = allo.customize(foo, verbose=True)

print(s.module)
```

**Buggy output**
Error: 'NoneType' object is not callable
Traceback (most recent call last): Line: 25
│   24: def foo(x: int32[2, 3]) -> int32[2, 3]:                                                                                  │
│   25:     return allo.dsl.add(1, x)                                                                                            │
Source Code 

**Expected behavior**
customize function of adding scalar and tensor without Error

the result of `allo.dsl.add(x, 1)`:

```mlir
module {
  func.func @foo(%arg0: memref<2x3xi32>) -> memref<2x3xi32> attributes {itypes = "s", otypes = "s"} {
    %alloc = memref.alloc() : memref<i32>
    %c1_i32 = arith.constant 1 : i32
    linalg.fill ins(%c1_i32 : i32) outs(%alloc : memref<i32>)
    %alloc_0 = memref.alloc() : memref<2x3xi32>
    linalg.broadcast ins(%alloc : memref<i32>) outs(%alloc_0 : memref<2x3xi32>) dimensions = [0, 1] 
    %alloc_1 = memref.alloc() : memref<2x3xi32>
    linalg.add {op_name = "add_0"} ins(%arg0, %alloc_0 : memref<2x3xi32>, memref<2x3xi32>) outs(%alloc_1 : memref<2x3xi32>)
    return %alloc_1 : memref<2x3xi32>
  }
}
```

chhzh123, 2025-03-23:
Hi @zsiothsu, thanks for raising the issue. Can you use the normal `+` instead (which already handles automatic shape broadcasting)? Currently the `allo.dsl.add` is used for other purposes.

```python
import allo
from allo.ir.types import int32
import numpy as np

def foo(x: int32[2, 3]) -> int32[2, 3]:
    return x + 1

s = allo.customize(foo, verbose=True)
print(s.module)

mod = s.build()
x = np.random.randint(0, 64, (2, 3)).astype(np.int32)
allo_y = mod(x)
np.testing.assert_allclose(allo_y, x + 1, atol=1e-5)
print("PASSED!")

```

zsiothsu, 2025-03-24:
> Hi @zsiothsu, thanks for raising the issue. Can you use the normal `+` instead (which already handles automatic shape broadcasting)? Currently the `allo.dsl.add` is used for other purposes. [...]

Thanks for your quick reply! I found this issue by chance. no impact on functionality

感谢您的及时回复！我也只是偶然发现了这个问题，并不影响使用

PR #328 (2025-03-26): Fix allo.dsl typing rules

## Description ##
Fixes #322 

This PR fixes type inference rules for functions in `allo.dsl`, which can be handled as general binary ops with shape broadcasting.

## Checklist ##

Please make sure to review and check all of these items:
- [ ] PR's title starts with a category (e.g. [Bugfix], [IR], [Builder], etc)
- [ ] All changes have test coverage (It would be good to provide ~2 different test cases to test the robustness of your code)
- [ ] Pass the [formatting check](https://cornell-zhang.github.io/allo/developer/index.html#id1) locally
- [ ] Code is well-docume [... cut]
closed 2025-03-26 by chhzh123