cornell-zhang/allo #333: Supported Vitis version to successfully run the tests after installation

YiminGao0113, 2025-03-31:
Hi, I wanted to ask about the specific version of Vitis to pass the tests in the installation process. If I don't specify my vitis path I can pass all the tests but if I used my Vitis 2019.2 I'm unable to pass 16 of the tests which are shown here - 

[image]

Could you kindly let me know if this is a version issue. I'm targeting XDEVICE=xilinx_u280_xdma_201920_3. 

Thanks in advance!

chhzh123, 2025-04-02:
Hi @YiminGao0113, some test cases for the VHLS backend are outdated. If you can run [this tutorial](https://cornell-zhang.github.io/allo/gallery/tutorial_02_vhls.html) correctly, you can safely ignore these errors in the unit tests

YiminGao0113, 2025-04-02:
Thanks @chhzh123 for the information! I was able to run the tutorial successfully without any issues.

The reason I was trying to use the tests is that I wanted to run a simple GEMM kernel using a configurable systolic array in Allo—something like [int8_gemm.py](https://github.com/cornell-zhang/allo/blob/main/playground/int8_gemm.py)  which seems similar to some of the tests under test_systolic. 
However, I’m currently running into an issue with affine.load when trying to run int8_gemm.py: (I believe it’s the same error I got when running my own systolic array test)

[image]

Do you happen to know what might be causing this error? Or is there another systolic array template I could use to run some tests? Thanks!

chhzh123, 2025-04-03:
I see. It might be a bug in the lowering process. We'll figure out the root cause later.

As we are switching to the new dataflow interface, you can check out this new systolic array example and see if it works.
https://github.com/cornell-zhang/allo/blob/main/tests/dataflow/test_systolic.py

YiminGao0113, 2025-04-03:
Thanks for pointing me to the new example! I tried and it works well in software simulation, but when I added the tests for hw_emu/hw by adding: 
        C = np.zeros((M, N), dtype=np.float32)
        hw_mod = df.build(top, target="vitis_hls", mode="hw_emu", project="gemm.prj")
        hw_mod(A, B, C)
        np.testing.assert_allclose(C, np.dot(A, B), atol=1e-5)
        print("hw Passed!")

I encountered the following errors related to shared buffers (buf0, buf1, and buf2):

ERROR: [v++ 200-779] Non-shared array 'buf0' (/home/yg9bq/allo_program/gemm.prj/kernel.cpp:271) failed dataflow checking: it can only have a single reader and a single writer.
ERROR: [v++ 200-779] Non-shared array 'buf1' (/home/yg9bq/allo_program/gemm.prj/kernel.cpp:273) failed dataflow checking: it can only have a single reader and a single writer.
ERROR: [v++ 200-979] Variable 'buf2' (/home/yg9bq/allo_program/gemm.prj/kernel.cpp:275) failed dataflow checking: it can only be written in one process function.
ERROR: [v++ 200-779] Non-shared array 'buf2' (/home/yg9bq/allo_program/gemm.prj/kernel.cpp:275) failed dataflow checking: it can only have a single reader and a single writer.

[image]

I’d really appreciate it if you could take a look when you get a chance—would be great to get a systolic array demo running on hw_emu or hw!

chhzh123, 2025-04-08:
Hi @YiminGao0113, sorry for the wrong pointer. If you want to run the systolic array on-board, please use this file:
https://github.com/cornell-zhang/allo/blob/main/tests/dataflow/test_daisy_chain_gemm.py

You can set the mode to "hw_emu" or "hw". This design mimics the one described in [AutoSA](https://autosa.readthedocs.io/en/latest/tutorials/optimize_array.html).

YiminGao0113, 2025-04-08:
I just successfully the design on my end and works correctly! Thank you so much!🙂

PR #338 (2025-04-10): [Fix] Fix legacy systolic array tests

## Description ##
This PR fixes #333.
1. Allow users to directly specify the buffer (using string) for partition
2. Fix some csyn/hw_emu incorrect number of arguments issue

## Checklist ##

Please make sure to review and check all of these items:
- [x] PR's title starts with a category (e.g. [Bugfix], [IR], [Builder], etc)
- [x] All changes have test coverage (It would be good to provide ~2 different test cases to test the robustness of your code)
- [x] Pass the [formatting check](https://cornell-zhang.github.io/allo/developer/index.html#id1) locally
- [ [... cut]
closed 2025-04-10 by chhzh123