cornell-zhang/allo #334: [BUG] incorrect lower 32-bit values in 2d array operations with bit slicing and bitcast operation

slbioig, 2025-04-03:
**Describe the bug**
When performing bit operations in Allo, there appears to be a bug related to the order of 1D and 2D array parameters when using bitcast and bit slicing operations. The lower 32 bits of the output are incorrectly set to 0 when a 2D array is declared before a 1D array, but the issue is resolved when the 1D array is declared before the 2D array.

```
def slicing_issue(inp: "Int(64)[10, 10]", A: "float32[10]") -> "Int(64)[10, 10]”:
    outp_data_pack: Int(64) 
    outp: Int(64)[10, 10] = 0
    data_temp_f : float32
    for i in range(10):
        for j in range(10):
            outp_data_pack = 0  
            data_temp_f = 1
            outp_data_pack[0:32] = data_temp_f.bitcast()
            outp_data_pack[32:64] = data_temp_f.bitcast()

            outp[i, j] = outp_data_pack

    return outp
```

```
def slicing_issue(A: "float32[10]", inp: "Int(64)[10, 10]") -> "Int(64)[10, 10]":
    
    outp_data_pack: Int(64) 
    outp: Int(64)[10, 10] = 0
    data_temp_f : float32
    for i in range(10):
        for j in range(10):
            outp_data_pack = 0  
            data_temp_f = 1
            outp_data_pack[0:32] = data_temp_f.bitcast()
            outp_data_pack[32:64] = data_temp_f.bitcast()

            outp[i, j] = outp_data_pack

    return outp
```

The only difference between the first and second codes is the order of the parameters, and they are expected to produce all ones after unpacked to fp32. However, in the first code, the lower 32 bits are all zeros.

**To Reproduce**
conditions:
1-D array and 2-D array as parameters
bit cast and bit slicing operation 
Order of array parameters
worked fine in csim mode only occurs in sw_emu, hw_emu, hw mode

**Buggy output**
output of the first code:
Value 0, 0: 0x3f80000000000000
  Lower 32-bit: 0x0
  Upper 32-bit: 0x3f800000
Value 0, 1: 0x3f80000000000000
  Lower 32-bit: 0x0
  Upper 32-bit: 0x3f800000
Value 0, 2: 0x3f80000000000000
  Lower 32-bit: 0x0
  Upper 32-bit: 0x3f800000
Value 0, 3: 0x3f80000000000000
  Lower 32-bit: 0x0
  Upper 32-bit: 0x3f800000
Value 0, 4: 0x3f80000000000000
  Lower 32-bit: 0x0
  Upper 32-bit: 0x3f800000

output of the second code
Value 0, 0: 0x3f8000003f800000
  Lower 32-bit: 0x3f800000
  Upper 32-bit: 0x3f800000
Value 0, 1: 0x3f8000003f800000
  Lower 32-bit: 0x3f800000
  Upper 32-bit: 0x3f800000
Value 0, 2: 0x3f8000003f800000
  Lower 32-bit: 0x3f800000
  Upper 32-bit: 0x3f800000
Value 0, 3: 0x3f8000003f800000
  Lower 32-bit: 0x3f800000
  Upper 32-bit: 0x3f800000
Value 0, 4: 0x3f8000003f800000
  Lower 32-bit: 0x3f800000
  Upper 32-bit: 0x3f800000

chhzh123, 2025-04-08:
Hi @slbioig, thanks for raising the issue! I suspect this is a bug when transferring parameters in NumPy/Python to the MLIR module. Could you also provide the test code you used?

slbioig, 2025-04-14:
[slicing_issue.txt](https://github.com/user-attachments/files/19738459/slicing_issue.txt)

zzzDavid, 2025-04-16:
Thank you for the complete test file! I confirm that I can reproduce this issue with `sw_emu` backend. I have created the same test for LLVM backend as follows, and I can confirm the LLVM backend does not have this issue. 

<details>
<summary>LLVM backend test</summary>

```python
import allo
from allo.ir.types import int32, float32, Int, Float, UInt
import numpy as np

# Type definition for 64-bit packing
io_pack_int64 = Int(64)

def slicing_issue_bug(inp: "Int(64)[10, 10]", A: "float32[10]") -> "Int(64)[10, 10]":
    
    outp_data_pack: Int(64) 
    outp: Int(64)[10, 10] = 0
    data_temp_f : float32
    for i in range(10):
        for j in range(10):
            outp_data_pack = 0  # Initialize only
            data_temp_f = 1
            outp_data_pack[0:32] = data_temp_f.bitcast()
            outp_data_pack[32:64] = data_temp_f.bitcast()

            outp[i, j] = outp_data_pack

    return outp
    
def slicing_issue_correct(A: "float32[10]", inp: "Int(64)[10, 10]") -> "Int(64)[10, 10]":
    
    outp_data_pack: Int(64) 
    outp: Int(64)[10, 10] = 0
    data_temp_f : float32
    for i in range(10):
        for j in range(10):
            outp_data_pack = 0  # Initialize only
            data_temp_f = 1
            outp_data_pack[0:32] = data_temp_f.bitcast()
            outp_data_pack[32:64] = data_temp_f.bitcast()

            outp[i, j] = outp_data_pack

    return outp

# NumPy Input & Build Configuration (32-bit packing)
np_dtype_inp = np.dtype("int64")  # Changed to 64-bit input
np_dtype_outp = np.dtype("int64")  # Changed to 64-bit output

# Step 1: Initialize with float32
inp_raw_32 = np.random.uniform(-3, 3, size=(10, 10)).astype(np.float32)
A = np.random.uniform(0.5, 1.5, size=(10)).astype(np.float32)

# Step 2: Convert float32 to uint32
inp_raw_32_int = inp_raw_32.view(np.int32)

# Step 3: Pack uint32 into 64-bit
inp_packed = np.zeros((10, 10), dtype=np.int64)
for i in range(10):
    for j in range(10):
        # Combine two uint32 into one uint64
        packed_value = 0
        packed_value += (inp_raw_32_int[i, j].astype(np.int64) << 0)   # Lower 32 bits
        packed_value += (inp_raw_32_int[i, j].astype(np.int64) << 32)  # Upper 32 bits
        
        inp_packed[i, j] = packed_value

# Check packed values
print("\n=== Packed Values Check ===")
print("First packed value:", inp_packed[0, 0])
[... 54 lines cut ...]
print("\n=== Unpacked Output Values (32-bit) - slicing_issue_bug ===")
print("Lower 32-bit values:")
print("Shape:", outp_float_32_low_bug.shape)
print("First 10 values:", outp_float_32_low_bug[:10, :10])
print("\nUpper 32-bit values:")
print("Shape:", outp_float_32_high_bug.shape)
print("First 10 values:", outp_float_32_high_bug[:10, :10])

print("\n=== Unpacked Output Values (32-bit) - slicing_issue_correct ===")
print("Lower 32-bit values:")
print("Shape:", outp_float_32_low_correct.shape)
print("First 10 values:", outp_float_32_low_correct[:10, :10])
print("\nUpper 32-bit values:")
print("Shape:", outp_float_32_high_correct.shape)
print("First 10 values:", outp_float_32_high_correct[:10, :10])
```
</details>

This is an excellent test case because the two input arguments `inp` and `A` are unused. This strongly suggests that it's an issue with OpenCL's runtime memory alignment for adjacent arguments with different data types. In OpenCL kernels, the arrangement of arguments can sometimes impact memory access patterns and optimizations that the compiler applies, which appears to be what's happening here.

I did another experiment where I add a "buffer" input with the same data type as the output between A and output: 
```
def slicing_issue_bug(inp: "Int(64)[10, 10]", A: "float32[10]", B: "Int(64)[10]") -> "Int(64)[10, 10]":
```
This produces correct bit packing results, which suggests it's an issue with memory alignment with OpenCL runtime.