cornell-zhang/allo #342: [Feature] Support function calls as data array indices

Oh-Afterglow, 2025-04-15:
**Is your feature request related to a problem? Please describe.**
Calculating array indices through function calls will help enable more complex array accessing patterns. For example, the per-stage data reordering in the butterfly network of [FEATHER](http://arxiv.org/abs/2405.13170) can be described as array accesses.

chhzh123, 2025-04-15:
@Oh-Afterglow Could you help create a minimal test case?

Oh-Afterglow, 2025-04-15:
A simple case modified from the producer-consumer example:

``` python
import allo
from allo.ir.types import float32
import allo.dataflow as df
import numpy as np

Ty = float32
M, N = 4, 4

def index_calculation(index: int):
    return index - 1

@df.region()
def top():
    pipe = df.pipe(dtype=Ty, shape=(), depth=4)

    @df.kernel(mapping=[1])
    def producer(A: Ty[M, N]):
        for i, j in allo.grid(M, N):
            # load data
            out: Ty = 0
            if j > 1:
                out = A[i, index_calculation(j)]
            else:
                out = A[i, j]
            # send data
            pipe.put(out)

    @df.kernel(mapping=[1])
    def consumer(B: Ty[M, N]):
        for i, j in allo.grid(M, N):
            # receive data
            data = pipe.get()
            # computation
            B[i, j] = data + 1

def test_producer_consumer():
    A = np.random.rand(M, N).astype(np.float32)
    B = np.zeros((M, N), dtype=np.float32)

    sim_mod = df.build(top, target="simulator")
    sim_mod(A, B)

if __name__ == "__main__":
    test_producer_consumer()
```

Currently building this kernel will trigger an error:
```
Error: 'index_calculation(j)' is not in list
```

chhzh123, 2025-04-16:
Notice nested functions should also use Allo types for type hint, so the `index_calculation` function should be:

```python
from allo.ir.types import index
def index_calculation(x: index) -> index:
    res: index = x - 1
    return res
```

The reason why here we use `index` type is because the loop variables and the array indices should both be `index`.

PR #345 (2025-04-16): [Dataflow] Support function calls in dataflow region

## Description ##
This PR fixes #342. It supports nested functions in the dataflow region. The dataflow functions are labeled as `df.kernel` using attributes.

### Examples ###
```python
def test_func_index():
    Ty = float32
    M, N = 4, 4

    def index_calculation(x: index) -> index:
        res: index = x - 1
        return res

    @df.region()
    def top():
        pipe = df.pipe(dtype=Ty, shape=(), depth=4)

        @df.kernel(mapping=[1])
        def producer(A: Ty[M, N]):
            for i, j in allo.grid(M, N):
                 [... cut]
closed 2025-04-16 by chhzh123