cornell-zhang/allo #347: [BUG] Testing a linear layer with systolic array on hardware

YiminGao0113, 2025-04-16:
**Describe the bug**
Hi @chhzh123 , thanks for helping with my previous issue on the systolic array testing on hardware. I wanted to test the linear layer on hardware as well and wanted to ask if the new systolic array is going to be updated to the allo.library? 
Below is my code (originally from tests/test_nn.py and I added the hw_emu test) to try to test the linear layer using the current systolic in the library using hw_emu but it failed and gives me a few errors. (I assume this could be related to the elder version of systolic array raising some issues but please correct me if I'm wrong)

**To Reproduce**
`# Copyright Allo authors. All Rights Reserved.
# SPDX-License-Identifier: Apache-2.0

import numpy as np
import allo
import allo.library.nn as nn
from allo.ir.types import int8, float32
import pytest

import allo.backend.hls as hls

def test_linear():
    from allo.library.systolic import systolic, systolic_tile

    # L, D = 512, 768
    # M0, M1 = 16, 16
    L, D = 8, 8
    M0, M1 = 2, 2
    W_A = np.random.randint(-4, 4, size=(D, 4 * D)).astype(np.int8)
    allo_C = np.zeros((L, 4 * D), dtype=np.int8)

    s = allo.customize(systolic, instantiate=[int8, int8, int8, L, D, 4 * D, M0, M1])
    # CPU testing
    mod = s.build()
    X = np.random.randint(-4, 4, size=(L, D)).astype(np.int8)
    mod(X, W_A, allo_C)
    np_C = X @ W_A
    np.testing.assert_allclose(allo_C, np_C, atol=1e-3)
    print("Passed!")
    # print(s.build(target="vhls"))

    
    if hls.is_available("vitis_hls"):
        # HW emulation testing
        mod_hw = s.build(target="vitis_hls", mode="hw_emu", project="test_linear_hwemu")
        C_hw = np.zeros((L, 4 * D), dtype=np.int8)
        mod_hw(X, W_A, C_hw)
        np.testing.assert_allclose(C_hw, np_C, atol=1e-3)
        print("HW Emulation test passed!")

if __name__ == "__main__":
    # pytest.main([__file__])
    test_linear()
`

**Buggy output**
`(allo) yg9bq@crabtree:/localsdd/yg9bq/allo_program$ python3 test_nn.py 
Passed!
LICENSE file not found
LICENSE file not found
LICENSE file not found
LICENSE file not found
LICENSE file not found
cd test_linear_hwemu; make run TARGET=hw_emu PLATFORM=$XDEVICE
g++ -o systolic xcl2.cpp host.cpp -I/opt/xilinx/xrt/include -I/localsde/all/Xilinx/Vivado/2022.2/include -Wall -O0 -g -std=c++1y -I. -fmessage-length=0 -L/opt/xilinx/xrt/lib -pthread -lOpenCL -lrt -lstdc++ 
mkdir -p ./_x.hw_emu.xilinx_u280_gen3x16_xdma_1_202211_1
v++ -c -g --save-temps  -t hw_emu --platform xilinx_u280_gen3x16_xdma_1_202211_1 -k systolic --temp_dir ./_x.hw_emu.xilinx_u280_gen3x16_xdma_1_202211_1  -I'.' -o'_x.hw_emu.xilinx_u280_gen3x16_xdma_1_202211_1/systolic.xo' kernel.cpp
Option Map File Used: '/localsde/all/Xilinx/Vitis/2022.2/data/vitis/vpp/optMap.xml'

****** v++ v2022.2 (64-bit)
  **** SW Build 3671529 on 2022-10-13-17:52:11
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.

INFO: [v++ 60-1306] Additional information associated with this v++ compile can be found at:
        Reports: /localsdd/yg9bq/allo_program/test_linear_hwemu/_x.hw_emu.xilinx_u280_gen3x16_xdma_1_202211_1/reports/systolic
[... 23 lines cut ...]
ERROR: [v++ 60-773] In '/localsdd/yg9bq/allo_program/test_linear_hwemu/_x.hw_emu.xilinx_u280_gen3x16_xdma_1_202211_1/systolic/systolic/vitis_hls.log', caught Tcl error: ERROR: [HLS 207-3746] subscripted value is not an array, pointer, or vector (/localsdd/yg9bq/allo_program/test_linear_hwemu/kernel.cpp:69:26)
ERROR: [v++ 60-773] In '/localsdd/yg9bq/allo_program/test_linear_hwemu/_x.hw_emu.xilinx_u280_gen3x16_xdma_1_202211_1/systolic/systolic/vitis_hls.log', caught Tcl error: ERROR: [HLS 207-3746] subscripted value is not an array, pointer, or vector (/localsdd/yg9bq/allo_program/test_linear_hwemu/kernel.cpp:73:27)
ERROR: [v++ 60-773] In '/localsdd/yg9bq/allo_program/test_linear_hwemu/_x.hw_emu.xilinx_u280_gen3x16_xdma_1_202211_1/systolic/systolic/vitis_hls.log', caught Tcl error: ERROR: [HLS 207-2095] array initializer must be an initializer list or string literal (/localsdd/yg9bq/allo_program/test_linear_hwemu/kernel.cpp:79:14)
ERROR: [v++ 60-773] In '/localsdd/yg9bq/allo_program/test_linear_hwemu/_x.hw_emu.xilinx_u280_gen3x16_xdma_1_202211_1/systolic/systolic/vitis_hls.log', caught Tcl error: ERROR: [HLS 207-2095] array initializer must be an initializer list or string literal (/localsdd/yg9bq/allo_program/test_linear_hwemu/kernel.cpp:80:14)
ERROR: [v++ 60-773] In '/localsdd/yg9bq/allo_program/test_linear_hwemu/_x.hw_emu.xilinx_u280_gen3x16_xdma_1_202211_1/systolic/systolic/vitis_hls.log', caught Tcl error: ERROR: [HLS 207-2095] array initializer must be an initializer list or string literal (/localsdd/yg9bq/allo_program/test_linear_hwemu/kernel.cpp:84:14)
ERROR: [v++ 60-773] In '/localsdd/yg9bq/allo_program/test_linear_hwemu/_x.hw_emu.xilinx_u280_gen3x16_xdma_1_202211_1/systolic/systolic/vitis_hls.log', caught Tcl error: ERROR: [HLS 207-2095] array initializer must be an initializer list or string literal (/localsdd/yg9bq/allo_program/test_linear_hwemu/kernel.cpp:88:14)
ERROR: [v++ 60-773] In '/localsdd/yg9bq/allo_program/test_linear_hwemu/_x.hw_emu.xilinx_u280_gen3x16_xdma_1_202211_1/systolic/systolic/vitis_hls.log', caught Tcl error: ERROR: [HLS 207-3776] use of undeclared identifier 'v24' (/localsdd/yg9bq/allo_program/test_linear_hwemu/kernel.cpp:148:34)
ERROR: [v++ 60-773] In '/localsdd/yg9bq/allo_program/test_linear_hwemu/_x.hw_emu.xilinx_u280_gen3x16_xdma_1_202211_1/systolic/systolic/vitis_hls.log', caught Tcl error: ERROR: [HLS 207-3776] use of undeclared identifier 'v25' (/localsdd/yg9bq/allo_program/test_linear_hwemu/kernel.cpp:149:34)
ERROR: [v++ 60-773] In '/localsdd/yg9bq/allo_program/test_linear_hwemu/_x.hw_emu.xilinx_u280_gen3x16_xdma_1_202211_1/systolic/systolic/vitis_hls.log', caught Tcl error: ERROR: [HLS 207-3776] use of undeclared identifier 'v26' (/localsdd/yg9bq/allo_program/test_linear_hwemu/kernel.cpp:150:34)
ERROR: [v++ 60-773] In '/localsdd/yg9bq/allo_program/test_linear_hwemu/_x.hw_emu.xilinx_u280_gen3x16_xdma_1_202211_1/systolic/systolic/vitis_hls.log', caught Tcl error: ERROR: [HLS 207-3339] no matching function for call to 'systolic_tile' (/localsdd/yg9bq/allo_program/test_linear_hwemu/kernel.cpp:180:7)
ERROR: [v++ 60-599] Kernel compilation failed to complete
ERROR: [v++ 60-592] Failed to finish compilation
INFO: [v++ 60-1653] Closing dispatch client.
makefile_us_alveo.mk:94: recipe for target '_x.hw_emu.xilinx_u280_gen3x16_xdma_1_202211_1/systolic.xo' failed
make: *** [_x.hw_emu.xilinx_u280_gen3x16_xdma_1_202211_1/systolic.xo] Error 1
Traceback (most recent call last):
  File "/localsdd/yg9bq/allo_program/test_nn.py", line 44, in <module>
    test_linear()
  File "/localsdd/yg9bq/allo_program/test_nn.py", line 37, in test_linear
    mod_hw(X, W_A, C_hw)
  File "/localsdd/yg9bq/allo/allo/backend/hls.py", line 429, in __call__
    raise RuntimeError("Failed to build the project")
RuntimeError: Failed to build the project`

Thanks in advance!

chhzh123, 2025-04-20:
Hi @YiminGao0113, as in the error message in your screenshot, you need to provide inputs and outputs as arguments to the generated module, i.e., `mod(X, allo_C)`, to run hw_emu successfully.

If you only want to run the GPT2 model, you can use [this PyTorch frontend](https://github.com/cornell-zhang/allo/blob/main/examples/torch/gpt2.py). Although it is not fully optimized, it can directly generate an executable hardware design.

For tiling large matrices, you can refer to [this](https://github.com/cornell-zhang/allo/blob/main/tests/dataflow/test_multi_cache_gemm.py) systolic array design and see if it works.

YiminGao0113, 2025-04-21:
Thanks! I just successfully tested the design and the GPT2 model with hw_emu mode.
One more question - do you have a floating-point systolic array template somewhere that I could test? 
Thanks!

chhzh123, 2025-04-22:
You can replace all the `int32` types in the systolic implementation with `float32`. However, it cannot guarantee outputting one element per cycle, which may potentially make the design stuck.

closed 2025-04-24 by YiminGao0113