cornell-zhang/allo #348: [Feature] Converting Top function parameters and return value into streaming buffers in Allo

slbioig, 2025-04-19:
**Is your feature request related to a problem? Please describe.**

When performing on-board execution in Allo, functions such as load_buf and store_res are added at the HLS level to read/write values on the global memory. During this process, an array of the same size as the input is created, all input values are stored in the input array, and then the subsequent computations are executed. Similarly, for the output, values are written to the global memory only after the return array is full.

This approach would be disadvantages in terms of on-chip memory and latency. To address this, I would like to convert the input/output arrays into streaming buffers, enabling overlapping operations where subsequent computations can proceed without waiting for the complete loading of the input, and data can be written to the global memory even if the output is not fully produced.

However, when attempting to apply the .to() schedule to the input parameters of the Top function at the Python level in Allo, a segmentation fault occurs. Additionally, for the return values, it is not possible to specify the destination (which would be the output array in the global buffer), making it impossible to convert the input/output arrays into streaming buffers on python level.

**Describe the solution you'd like**

I'm not sure whether this is unsupported or a bug, but applying .to() to the input parameter causes a segmentation fault. Additionally, it's not possible to specify the destination in the return value, making it impossible to convert the input/output array into a stream. I would greatly appreciate it if an update could be made to allow the input parameter and return value to be converted into streaming buffers.

**Additional context**

I created an example using the test_nested_function example from the test_schedule_stream.py file and checked the waveform.
The Allo code is as follows:

```
T=float32
def func1(A: T[10, 20], B: T[10, 20]):
    for i, j in allo.grid(10, 20):
        B[i, j] = A[i, j] + 1

def func2(A: T[10, 20], B: T[10, 20]):
    for i, j in allo.grid(10, 20):
        B[i, j] = A[i, j] * 2

def top(A: T[10, 20]) -> T[10, 20]:
    B: T[10, 20]
    C: T[10, 20]
    func1(A, B)
    func2(B, C)
    return C

s = allo.customize(top)
s.to(s.B, "func2", depth=1)
```

The generated HLS top function is as follows.

```
void top(
  float *v22,
  float *v23
) {	// L26
  #pragma HLS interface m_axi port=v22 offset=slave bundle=gmem0
  #pragma HLS interface m_axi port=v23 offset=slave bundle=gmem1
  #pragma HLS dataflow
  float buf0[10][20];	//
  load_buf0(v22, buf0);	//
  hls::stream< float > B /* B[10][20] */;	// L27
  #pragma HLS stream variable=B depth=2
  float C[10][20];	// L28
  func1(buf0, B);	// L29
  func2(B, C);	// L30
  store_res1(C, v23);	//
}
```
The load_buf0 function stores the entire input array in the created buf0, and when C is fully filled, the store_res1 function writes the values to the global memory.

On the waveform,

[image]

you can see that the read/write operations with the global memory and the kernel's operation do not overlap.

Next is the code where the input/output array is changed to a streaming buffer. This can be easily modified at the HLS level.

```
void load_buf0(
  float v12[200],
  hls::stream< float > &v13 /* v13[10][20] */
) {	//
  l_S_load_buf0_load_buf0_l_0: for (int load_buf0_l_0 = 0; load_buf0_l_0 < 10; load_buf0_l_0++) {	//
    l_load_buf0_l_1: for (int load_buf0_l_1 = 0; load_buf0_l_1 < 20; load_buf0_l_1++) {	//
    #pragma HLS pipeline II=1 rewind
      float v16 = v12[((load_buf0_l_0 * 20) + load_buf0_l_1)];	//
      v13.write(v16);	//
    }
  }
}

void store_res1(
  hls::stream< float > &v17,
  float v18[200]
) {	//
  #pragma HLS stream variable=v17 depth=2
  l_S_store_res1_store_res1_l_0: for (int store_res1_l_0 = 0; store_res1_l_0 < 10; store_res1_l_0++) {	//
    l_store_res1_l_1: for (int store_res1_l_1 = 0; store_res1_l_1 < 20; store_res1_l_1++) {	//
    #pragma HLS pipeline II=1 rewind
      float v21 = v17.read();
      v18[((store_res1_l_0 * 20) + store_res1_l_1)] = v21;	//
    }
  }
}

void top( float *v22,
  float *v23
) {	// L26
  #pragma HLS interface m_axi port=v22 offset=slave bundle=gmem0
  #pragma HLS interface m_axi port=v23 offset=slave bundle=gmem1
  #pragma HLS dataflow
  hls::stream< float > buf0/* buf0[10][20] */;	// L27
  load_buf0(v22, buf0);	//
  hls::stream< float > B/* B[10][20] */;	// L27
  #pragma HLS stream variable=B depth=2
  hls::stream< float > C;
  #pragma HLS stream variable=B depth=2
  func1(buf0, B);	// L29
  func2(B, C);	// L30
  store_res1(C, v23);	//
}
```
Simply declared buf0 and C as has::stream and changed the parameters and operators in load_buf0 and store_res1.

When checking the waveform in this code, you can see that the read/write operations in global memory and the kernel operations overlap.

[image]

Although it is a simple modification at the HLS level, I am not sure how difficult it would be to implement at the Python/MLIR level. However, since this issue directly affects the performance of onboard execution, I hope it will be implemented and I would greatly appreciate your efforts.

[example_no_stream.tar.gz](https://github.com/user-attachments/files/19818575/example_no_stream.tar.gz)
[example_stream.tar.gz](https://github.com/user-attachments/files/19818576/example_stream.tar.gz)

The project folder to check the waveform is attached. You can verify the waveform by running hardware emulation.

yk-choi-yk, 2025-04-19:
A related question - is there any option to avoid generation of "load_buf"s and "store_res"s ? 
For example in @slbioig 's case, I think A[i, j] reference directly be translated to float* v22 reference in the func1 without the need for a load_buf.

chhzh123, 2025-04-20:
@yk-choi-yk @slbioig Thanks for pointing this out! Yes, you can turn off the load/store buffer generation by setting `wrap_io` as `False` in the `s.build` function. However, it may not work very well for multi-dimensional arrays, as it requires flattening all the memory accesses.

We did have some streaming buffer support for the top-level function in our previous implementation, but it could not work well for more complicated cases where the buffers need to be read multiple times (e.g., reduction in GEMM), so we are still figuring out a way to systematically analyze it.

There are two workarounds now:
1. Manually modified the Allo-generated HLS code just like what you did in the example, and integrate it into the Python testing loop using [`allo.IPModule`](https://cornell-zhang.github.io/allo/dive/ip.html).
2. We are developing a new dataflow interface to better handle these cases. You could check out the [producer-consumer](https://github.com/cornell-zhang/allo/blob/main/tests/dataflow/test_producer_consumer.py) or the [systolic array](https://github.com/cornell-zhang/allo/blob/main/tests/dataflow/test_systolic.py) cases here. Basically we explicitly expose a stream type to the users, so users can better communicate between different kernels using stream.

Please let me know if you have any further questions.